OpenHD/OpenHD #953: Air unit can't receive packet from ground unit and got a lot loopback packets

afilez, 2023-08-24:
**Describe the bug**
Running Open.HD  air unit on OrangePi H3, ath9k_htc WiFi adapter.
Ground unit running on ubuntu X86 PC station.
Video didplay is fine but click any fetch air or fetch ari camera will raise timeout.

after turn the options on :
bool advanced_debugging_rx = true;
bool advanced_latency_debugging_rx= true;

I'd checked the air unit console and found :

...
[1970-01-01 00:08:58.696] [WBTxRx] [warning] Discarding packet due to pcap parsing error!
[1970-01-01 00:08:58.696] [WBTxRx] [debug] Got packet 0 257
[1970-01-01 00:08:58.696] [WBTxRx] [warning] Discarding packet due to pcap parsing error!
[1970-01-01 00:08:58.696] [WBTxRx] [debug] Got packet 0 318
[1970-01-01 00:08:58.696] [WBTxRx] [warning] Discarding packet due to pcap parsing error!
[1970-01-01 00:08:58.696] [WBTxRx] [debug] Got packet 0 318
[1970-01-01 00:08:58.696] [WBTxRx] [warning] Discarding packet due to pcap parsing error!
[1970-01-01 00:08:58.696] [WBTxRx] [debug] Got packet 0 68
[1970-01-01 00:08:58.696] [WBTxRx] [warning] Discarding packet due to pcap parsing error!
[1970-01-01 00:08:58.697] [WBTxRx] [debug] Got packet 0 68
...

It looks the wbtxrx got a lot of invalid packets from the ‘pcap_nex’ function and parse them as nothing.

**Hardware:**
 - Air side 
 -- Pi model
OrangePi Zero Pluse2 H3
 -- Camera(s)
Withou camera and tried some UVC cameras.
 -- Wifi card(s)
ath9khtc, AP6212(brcmfmac)
 -- Other hardware
 - Ground side
 -- Pi model
Ubuntu 22.04 and Ubuntu 20.04
 -- Wifi card(s)
ath9k_htc
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?
not using any RC and FC yet.

Other:
Running on mainline Linux-kernels and WiFi adapters drivers, same lippcap version, same firmware on the ground and air units.
and they works prety well on the x86 pc linux(s) , included kernl 3.10 on Centos7 ,.
buildroot to create the taget system image.

BTW, I guess Open.HD make as single execute appclation is qute good, but  ... the scripts is TOOO friendly and not fit to me, all I need just.. automake or cmake, sure, now 2.5-evo is greater then the other.

Consti10, 2023-08-28:
Hello,
Sorry for the late answer, i just saw your issue.

First up - I highly recommend you rtl8812au or rtl8812bu. Better driver, 5G and 2.4G depending which card you buy, and in the next release, we'l actually deprecate AR9271 - mostly because its driver has such less features than 8812 and it is also a bit bugged -
For example, i think the issue you see with the incorect packets is the driver reporting injected packets back as received
(that is a known bug of AR)

However, pcap parsing error hints at an invalid radiotap header - perhaps kernel 3.X and its radiotap header is incompatible with what we currently use (5.X or 6.X). Doesn't mean you cannot use kernel 3.X, but you probably need to figure out the parsing error yourself, since i don't have such a device for testing.

regarding building - well, just run mkdir build & cmake .. & make  & make install (optionally)
I think cmake can do pretty much everything ;)

afilez, 2023-08-29:
Thanks, so I got some rtl8812au cards now, let me take more try.  Just confuse how it happend, maybe I can find it out later.

closed 2023-08-29 by afilez