OpenHD/OpenHD #996: OpenHDImageWriter depends upon debian pacakges not requested

bmsleight, 2023-12-30:
OpenHDImageWriter required some QT packages when used with Debian Bokworm. 

bms@tartt:~/Downloads$ OpenHDImageWriter 
QQmlApplicationEngine failed to load component
qrc:/main.qml:10:1: module "QtQuick.Controls.Material" is not installed
qrc:/main.qml:9:1: module "QtQuick.Layouts" is not installed
qrc:/main.qml:8:1: module "QtQuick.Controls" is not installed
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
qrc:/main.qml:6:1: module "QtQuick" is not installed
qrc:/main.qml:10:1: module "QtQuick.Controls.Material" is not installed
qrc:/main.qml:9:1: module "QtQuick.Layouts" is not installed
qrc:/main.qml:8:1: module "QtQuick.Controls" is not installed
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
qrc:/main.qml:6:1: module "QtQuick" is not installed
qrc:/main.qml:10:1: module "QtQuick.Controls.Material" is not installed
qrc:/main.qml:9:1: module "QtQuick.Layouts" is not installed
qrc:/main.qml:8:1: module "QtQuick.Controls" is not installed
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
qrc:/main.qml:6:1: module "QtQuick" is not installed
qrc:/main.qml:10:1: module "QtQuick.Controls.Material" is not installed
qrc:/main.qml:9:1: module "QtQuick.Layouts" is not installed
qrc:/main.qml:8:1: module "QtQuick.Controls" is not installed
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
qrc:/main.qml:6:1: module "QtQuick" is not installed
qrc:/main.qml:10:1: module "QtQuick.Controls.Material" is not installed
qrc:/main.qml:9:1: module "QtQuick.Layouts" is not installed
qrc:/main.qml:8:1: module "QtQuick.Controls" is not installed
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
qrc:/main.qml:6:1: module "QtQuick" is not installed
bms@tartt:~/Downloads$ apt-cache search  qml-module-qtquick-controls2
qml-module-qtquick-controls2-suru - Suru style for QtQuick Controls 2
qml-module-qtquick-controls2 - Qt 5 Qt Quick Controls 2 QML module
bms@tartt:~/Downloads$ sudo apt-get install  qml-module-qtquick-controls2
[sudo] password for bms: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtquick-templates2
The following NEW packages will be installed:
  libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtquick-controls2 qml-module-qtquick-templates2
0 upgraded, 4 newly installed, 0 to remove and 118 not upgraded.
Need to get 1,433 kB of archives.
After this operation, 8,647 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://deb.debian.org/debian bookworm/main amd64 libqt5quicktemplates2-5 amd64 5.15.8+dfsg-2 [397 kB]
Get:2 http://deb.debian.org/debian bookworm/main amd64 libqt5quickcontrols2-5 amd64 5.15.8+dfsg-2 [62.7 kB]
Get:3 http://deb.debian.org/debian bookworm/main amd64 qml-module-qtquick-templates2 amd64 5.15.8+dfsg-2 [70.4 kB]
Get:4 http://deb.debian.org/debian bookworm/main amd64 qml-module-qtquick-controls2 amd64 5.15.8+dfsg-2 [903 kB]
Fetched 1,433 kB in 0s (3,742 kB/s)                   
Selecting previously unselected package libqt5quicktemplates2-5:amd64.
(Reading database ... 132605 files and directories currently installed.)
Preparing to unpack .../libqt5quicktemplates2-5_5.15.8+dfsg-2_amd64.deb ...
Unpacking libqt5quicktemplates2-5:amd64 (5.15.8+dfsg-2) ...
Selecting previously unselected package libqt5quickcontrols2-5:amd64.
Preparing to unpack .../libqt5quickcontrols2-5_5.15.8+dfsg-2_amd64.deb ...
Unpacking libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
Selecting previously unselected package qml-module-qtquick-templates2:amd64.
Preparing to unpack .../qml-module-qtquick-templates2_5.15.8+dfsg-2_amd64.deb ...
Unpacking qml-module-qtquick-templates2:amd64 (5.15.8+dfsg-2) ...
Selecting previously unselected package qml-module-qtquick-controls2:amd64.
Preparing to unpack .../qml-module-qtquick-controls2_5.15.8+dfsg-2_amd64.deb ...
Unpacking qml-module-qtquick-controls2:amd64 (5.15.8+dfsg-2) ...
Setting up libqt5quicktemplates2-5:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtquick-templates2:amd64 (5.15.8+dfsg-2) ...
Setting up libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtquick-controls2:amd64 (5.15.8+dfsg-2) ...
Processing triggers for libc-bin (2.36-9+deb12u3) ...
bms@tartt:~/Downloads$ OpenHDImageWriter 
QQmlApplicationEngine failed to load component
[... 233 lines cut ...]
Setting up libqt5qmlworkerscript5:amd64 (5.15.8+dfsg-3) ...
Setting up qml-module-qtquick2:amd64 (5.15.8+dfsg-3) ...
Processing triggers for libc-bin (2.36-9+deb12u3) ...
bms@tartt:~/Downloads$ OpenHDImageWriter 
QQmlApplicationEngine failed to load component
qrc:/main.qml:7:1: module "QtQuick.Window" is not installed
bms@tartt:~/Downloads$ ^C
bms@tartt:~/Downloads$ sudo apt-get install qml-module-qtquick-window2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  qml-module-qtquick-window2
0 upgraded, 1 newly installed, 0 to remove and 118 not upgraded.
Need to get 27.8 kB of archives.
After this operation, 134 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main amd64 qml-module-qtquick-window2 amd64 5.15.8+dfsg-3 [27.8 kB]
Fetched 27.8 kB in 0s (284 kB/s)                
Selecting previously unselected package qml-module-qtquick-window2:amd64.
(Reading database ... 133131 files and directories currently installed.)
Preparing to unpack .../qml-module-qtquick-window2_5.15.8+dfsg-3_amd64.deb ...
Unpacking qml-module-qtquick-window2:amd64 (5.15.8+dfsg-3) ...
Setting up qml-module-qtquick-window2:amd64 (5.15.8+dfsg-3) ...
bms@tartt:~/Downloads$ OpenHDImageWriter 
^C

raphaelscholle, 2024-01-02:
we do only support ubuntu jammy and ubuntu lunar, sadly qt is quite annoying when it comes to different distros, so I don't really have planned to change that

bmsleight, 2024-01-05:
Noted - thanks . The log show how I resolved in debian. Noted not supported.

raphaelscholle, 2024-01-09:
I can look into that in the next days and look if I can add the missing files without affecting the ubuntu installations, if it is possible I'll just do it, if not I'm sorry, when requested I could add debian support in the future

closed 2024-01-09 by raphaelscholle