OpenHD/OpenHD #1015: Wifi Hotspot  - main password can't be used. Can't connect

AlexGromSwarmly, 2024-03-14:
Can't connect to the wifi hotspot "openhd".

Hotspot created with WPS secure and in WPS password can be no longer than 8 chars.  
The default password is "openhdopenhd"  12 chars.

Please correct it!

Also: There is written tat Hotspot wil be off if Armed, So how can I have telemetry on my Tablet by WifiHotspot while flying?
Can it be optional switching on or off if Armed?

knputyi, 2024-03-14:
It's bad to have internal wifi active while flying. Can/could interfer with wifibroadcast OHD. Use a wired connection to your tablet, eg usb tethering or ethernet 2 usb adapter from pi to tablet.

AlexGromSwarmly, 2024-03-14:
Sorry, I do not agree.  Hotspot on 2.4 with 25mw will not interfere with 5.8 with 600mW so much.  And in order to be free from wiring and more mobile you should have hotspot.  Even to provide video on mobile phone. I use to do in in 2.0.12 OpenHD, and it was working perfect!

knputyi, 2024-03-14:
in Qopenhd app, ground tab, can you make the hotspot setting to be "on", not auto?

knputyi, 2024-03-14:
[image]

knputyi, 2024-03-14:
sure it can, I just connected my Samsung galaxy s21 to it

AlexGromSwarmly, 2024-03-14:
[image]
I have this on my laptop

knputyi, 2024-03-14:
Good. Now please, in the future contact us on telegram or discord, it's easier to help there.