OpenHD/OpenHD #1032: [Feature Request] support for the HTI ht-203u thermal camera

TechCom2020, 2024-04-07:
I thought the hti ht-203u might work already since it is an hti (which already has some supported models) and It works with a third party android app "usb camera" as do other generic cameras that also work with openhd, so i was thinking it might also work with openhd as well, but no luck.
This camera, while not as high resolution as the ht-301 currently supported, offers an incredible value in price vs. quality (256X192 Resolution) in my opinion seeing as it can be had on aliexpress for around $135.

5B4APV, 2024-05-18:
Would also like to see this camera supported as I was unable to get it to work.
Here is my post on the eevblog with information I got from it: https://www.eevblog.com/forum/thermal-imaging/hti-ht-203u-thermal-imager-raspberry-pi-os/

Krastanov, 2024-05-18:
With `qv4l2` it works enough to show an image, but it fails to obey any configuration requests. Run with `qv4l2 --device=/dev/videoX`; see available devices with `v4l2-ctl --list-devices`.

raphaelscholle, 2024-07-10:
with thermal cameras, we currently can only support cameras that we do have access to, or that "just work", We'd like to add more, but without the ability to test, and no funds to buy them, we can't just support everything.

TechCom2020, 2024-07-13:
Tried it with a TOOLTOP T7 camera (mentioned on discord I think) and it just works. It's about the same quality and price so I recommend this instead.

closed 2024-07-13 by TechCom2020