OpenHD/OpenHD #1073: Transmit antenna tracker position from ground station to drone

rsaxvc, 2024-05-13:
**Describe the enhancement**
1) Parse mavlink messages from the antenna tracker on the ground station.
2) Forward relevant packets to the air unit. Possibly with rate limiting. Packets that should be forwarded periodically include SYS_STATUS and GLOBAL_POSITION_INT.

**Expected behavior**
Once my ground station has a GPS fix, I expect that it will upload the ground station position over OpenHD's WiFi-ish link, so that it may be used by relevant modules on the vehicle.

**Additional context**
This would be used to upload control station position for transmission over remote ID module.

Consti10, 2024-05-22:
Can you hint me a explainer or link how this proposed functionality is used by any FC firmware ?
Normally, the home position is simply determined at arm atc, and an uplink from tracker to FC is not required.

rsaxvc, 2024-05-22:
It's not for the FC home position, it's for RemoteID (USA regulatory requirement) when that functionality is integrated as built.

https://www.ecfr.gov/current/title-14/chapter-I/subchapter-F/part-89

Depending on how RemoteID is integrated, it may need to transmit take off location, or it may need to transmit control-station location (which could be different or update).

The flight controller does have related responsibilities like refusing to take off until RemoteID transmission is working, but this ticket is just a request to upload ground control station position at a low rate when available.

Consti10, 2024-05-24:
Uff, this sounds quite specific.  I for myself neither have a RemoteID module nor tracker since I am located in EU, so let's see if someone is interested. Implementation should be quite straight forward, they key parts are:

1) Read data from the tracker serial (right now we only send data to it for simplicity)
2) optional, filter it's messages to not pollute the uplink
3) send those messages up

rsaxvc, 2024-05-24:
It seems EU may require something similar for some categories of drones. I haven't looked at the detailed requirements.

https://www.easa.europa.eu/en/document-library/general-publications/remote-identification-will-become-mandatory-drones-across

https://enterprise-insights.dji.com/blog/navigating-remote-id-compliance-for-drone-operations-in-the-eu

rsaxvc, 2025-03-23:
I was looking at the ASTM spec for RemoteID, and found that it also supports an 802.11 based transmission. I was able to mock up a transmission with a fixed payload, though it isn't always appearing in the OpenDroneID app yet. Would this be something OpenHD might be interested in integrating?

[image]

closed 2026-10-20 by raphaelscholle