OpenHD/OpenHD #1099: how to adjust the exposure of version 2.6 carefully

zznlzz, 2024-08-07:
In qopenhd, there are only five options -10/-5/0/5/10. What should I do if I want to adjust it to -2? In the early version, I can adjust it in openhd-settings.txt, but this is not available in the new version.

knputyi, 2024-08-07:
Can you connect a keyboard to ground pi and enter any value, manually?

knputyi, 2024-08-07:
Join our Telegram channel, you'll get help much faster.
https://t.me/OpenHD_User

zznlzz, 2024-08-08:
> 您可以将键盘连接到接地 pi 并手动输入任何值吗？

Yes, so do I need to enter the relevant libcamera command to set it on the screen after the ground-end system is turned on? Is there any way to keep the settings every time I start up like Qopenhd or modify the sd card parameters? Thank you for your reply.

raphaelscholle, 2024-08-12:
if you enable advanced settings, the value in the right sidebar gets transformed into a text-field and you can just set it to any value, not only in predefined steps. No need to do it with ssh or the command line

closed 2024-08-12 by raphaelscholle

zznlzz, 2024-08-13:
> 如果启用高级设置，右侧边栏中的值将转换为文本字段，您可以将其设置为任何值，而不仅仅是在预定义的步骤中。无需使用 ssh 或命令行执行此操作
thanks, but how to enable advanced settings?

raphaelscholle, 2024-08-13:
go to the openhd menu -> osd -> enable advanced settings (in older versions it's called show whitelisted settings and show advanced button), then just go to the camera tab and refresh, you'll see a lot more settings

zznlzz, 2024-08-15:
> go to the openhd menu -> osd -> enable advanced settings (in older versions it's called show whitelisted settings and show advanced button), then just go to the camera tab and refresh, you'll see a lot more settings

thanks again, i found it.

raphaelscholle, 2024-08-15:
no worries, if you need more in depth help, please visit us in our Telegram channel or discord. There we can help in a much shorter timeframe.

zznlzz, 2024-08-15:
> no worries, if you need more in depth help, please visit us in our Telegram channel or discord. There we can help in a much shorter timeframe.

[image]
If it's convenient, please take a look at my new issue on github, which has been bothering me for a long time.