OpenHD/OpenHD #1105: Version 2.6.0 may be disconnected

zznlzz, 2024-08-15:
# environment
Open HD version 2.6.0, raspberry pie 3B+

# problem
1 Mosaic interference (not stripes) will appear when connecting.
2 Every once in a while, the sky end will be disconnected from the ground end.

# The efforts that have been made
1 The network card and Raspberry Pie are powered separately, and the voltage stabilizing module is used.
2 The ground end uses dual network cards to receive.
3 Camera csi cable is wrapped with copper foil.
4 when changing the test site (indoor and outdoor and open space), there will be disconnection.
5 There is little difference between changing different Hz.

# Compared with version 2.0.8
I use the same wiring. When I use version 2.0.8, it will not be stuck, and the network cards on the ground side and the sky side will light up. When using version 2.6.0, I can see that the network card is recognized by both the sky end and the ground end with qopenhd, but the network card doesn't light up. I suspect that the jam and disconnection are related to this.

Droneforest05, 2024-08-15:
Hello zznlzz
Evo versions are more power hungry so maybe you have a problem in your power supply 
The led behavior is normal because the kernel is different and in this one the led is not controllable 
About the lines or mosaic maybe you can share a photo to be more sure about what do you mean
Happy flying

Droneforest05, 2024-08-15:
To test the power problem you should tried to fly at mc1 or 0 and at lower power
What tpi level are you using now?

zznlzz, 2024-08-15:
> 要测试功率问题，您应该尝试在 mc1 或 0 和较低功率下飞行：您现在使用的是哪个 tpi 级别？

I have tried tpi for low and high and the results are almost the same. I will try to solve the problem of insufficient power supply you mentioned. Thank you for your reply.

raphaelscholle, 2024-08-15:
also .. what cards are you using, what antennas, ...

zznlzz, 2024-09-04:
I solved this problem by replacing the buck module.

closed 2024-09-04 by zznlzz