OpenHD/OpenHD #1106: radxa zero 2 pro

sokeman, 2024-08-21:
radxa zero 2 pro is the fastest and by far the cheapest sbc can you please support it???

raphaelscholle, 2024-08-22:
The zero2 pro is not on the timeline and most likely will never get supported.
There are substantial differences between the radxa zero3w and 2 pro, which basically need a lot of work to implement.
Currently we do only support  SBC's with Broadcom,Rockchip,Allwinner,Qualcomm chipsets, there is no Pla to add Amlogic socs.
It's also very expensive to add air support to new soc's , so the investement is just to high

sokeman, 2024-08-22:
i dont understand the specifics but the radxa zero2 pro cost $100aud and one of the ones you suggested cost 1000. I would like to be able to add to the community, what do i have to learn to do that??  At the moment i know some programming, python and i started a full stack web development course?  ill send some money when i have it but im on disability so mostly i can just give my time!

thanks a lot for the email (usually people just ignore you)

tim

raphaelscholle, 2024-08-22:
the pi zero3w is a lot cheaper than that, while being supported. 
If you want to port openhd, on the zero2 pro, the task would be to modify the drm-kms driver to render without doing a double vsync, and with cameras, you can just commission them, usually they start in the 10.000 usd area, which makes it not that easy to get new ones on the boards. (https://arace.tech/products/radxa-zero-3w?_pos=7&_sid=4382229b0&_ss=r) starting at 15usd.
For the air, we're still struggeling with new cameras, since it's a lot of money and a lot of time needed, but there I'm pretty sure we'll add some in the near future.

sokeman, 2024-08-22:
i dont really know how to do that(drm-kms), i know its a bit of effort but could you possibly look on udemy and give me a url to a course i can lear(if they have one)??

thanks

raphaelscholle, 2024-08-22:
That's nothing you can learn from tutorials, and only something experienced people should attempt.

If you want to test things, you can see if you get video via kmssink and overlay openhd via eglfs, that would be the first step I'd do, if you don't find a way it's basically not possible at all.

sokeman, 2024-08-22:
is that with the radxa zero 2 pro or the radxa zero 3w??   ive got a 3w but i need a camera.

thanks tim

raphaelscholle, 2024-08-22:
the zero3w already works as ground, and doesn't work, for cameras, we're still working on it together with arducam

sokeman, 2024-08-22:
just checked, its a zero 2 but i can get a 3w version to test if that would help??

sokeman, 2024-08-22:
what size ram and emmc do you suggest for ground and air??

thanks tim

raphaelscholle, 2024-08-25:
sorry for the late reply, It would be a lot faster if we have that communication via telegram or discord.
For openhd the  smallest variants are perfectly usable,but for developing, I recommend at least 2gb

sokeman, 2024-08-25:
thats ok my username is "sokeman."  (with the dot)

sokeman, 2024-08-25:
"sokeman." with the dot for discord. are you talking about the orange pi 3??  because the zero only comes in 512mb.

raphaelscholle, 2024-08-25:
just join our official discord https://discord.gg/NRRn5ugrxH

I'm talking about the [radxa zero3w ](https://radxa.com/products/zeros/zero3w/)

closed 2026-09-14 by raphaelscholle