OpenHD/OpenHD #1111: Reduce openhd boot time

zznlzz, 2024-09-04:
I use OpenHD2.6.0. 
After the power is turned on, the ground terminal will wait for a long time in the tty1 login interface to get an image. How can we minimize the startup time?
[image]

raphaelscholle, 2024-09-04:
The biggest impact is actually storage speed, if you use usb3, emmc or a ssd to boot it is reduced significantly.
The rest is mostly the slow gpu of the pi, that's also why the osd is slowly getting drawn initially

zznlzz, 2024-09-05:
> 影响最大的其实是存储速度，如果使用 usb3、emmc 或者 ssd 来启动，速度会明显降低。其余的大部分是 pi 的慢速 gpu，这也是 osd 最初缓慢绘制的原因

I don't quite understand the meaning of "use usb3, emmc or a ssd to boot". In addition, will it be effective if I replace pi3b with pi4b by hardware

raphaelscholle, 2024-09-08:
Usually the rpi boots from sd-card, but with a cm4 you can also use emmc, or you can use a usb-stick to boot from, if those are fast you'll automatically reduce the starttime.

The pi4 also is a little faster, but I don't know if there is a real difference between pi3 and pi4

closed 2024-10-24 by zznlzz