OpenHD/OpenHD #1128: USB camera mjpg format cannot obtain the image

HEM-ai, 2024-11-29:
When using a USB camera on Windows, both yuy2 and mjpg formats can capture images normally. However, when using openHD and radxa development boards, mjpg format cannot capture images, and QGroundControl prompts that the camera is restarting
[image]

raphaelscholle, 2024-11-30:
openhd has no support for mjpeg

HEM-ai, 2024-12-02:
If I want to support the mjpeg, can I implement it by modifying the openHD source code? Is the modification complicated? What aspects can be modified?

raphaelscholle, 2024-12-03:
You could do it, with a few hours invested, however, you'd need to find a way to control the mjpeg-stream bitrate from the camera, within openhd, this might be impossible or very hard, needing special sdk's or things like that.If you don't do that, you'll not get a picture or extremely high loss, since the "usual 20Mbit-50Mbit" mjpeg doesn't fit in the link. you might have to reduce the quality a lot (if the encoder even allows for this).

HEM-ai, 2024-12-03:
thank you. It is possible to compile OpenHD with g++, but I want to cross compile OpenHD. Using gcc-arm-10.3-2021.07-x86_64-arch64-none Linux GNU always reports errors. Which compiler should I use for cross compilation

raphaelscholle, 2024-12-05:
we compile via qemu, so not directly crosscompiling

HEM-ai, 2024-12-06:
How is the Radxa image file created? Can you provide specific instructions

HEM-ai, 2024-12-06:
I replaced /usr/local/bin/openhd in the radxa development board with cross compiled binary openhd, but as a result, air doesn't work and I can't find the wifi openhd_air. Why is this

raphaelscholle, 2024-12-10:
OpenHD has a few libraries which need to be the correct version to work, which radxa board are you using ?

We use a qemu chroot environment to compile OpenHD and any needed app from within the official image from radxa, there we can make sure that all libs are the correct version and everything works

Most likely you have a library version missmatch

closed 2026-09-14 by raphaelscholle